aparcar/openwrt #14: DSA support

aparcar, 2021-09-29:
Supported targets:
- [ ] apm821xx
- [ ] arc770
- [ ] archs38
- [ ] armvirt
- [ ] at91
- [ ] ath25
- [ ] ath79
- [ ] bcm27xx
- [ ] bcm47xx
- [ ] bcm4908
- [ ] bcm53xx
- [ ] bcm63xx
- [ ] bmips
- [ ] gemini
- [ ] generic
- [ ] imx6
- [ ] ipq40xx
- [ ] ipq806x (wip: https://github.com/openwrt/openwrt/pull/4036)
- [ ] ipq807x
- [ ] kirkwood
- [ ] lantiq
  - [x] xrx200 (via: https://github.com/openwrt/openwrt/commit/b1df48caac1b3f75a65768a5abf8de84801edbf0)
- [ ] layerscape
- [ ] Makefile
- [ ] malta
- [ ] mediatek
- [ ] mpc85xx
- [ ] mvebu
- [ ] mxs
- [ ] octeon
- [ ] octeontx
- [ ] omap
- [ ] oxnas
- [ ] pistachio
- [ ] ramips
- [ ] realtek
- [ ] rockchip
- [ ] sunxi
- [ ] tegra
- [ ] uml
- [ ] x86
- [ ] zynq

Notes from @hauke:
> We will migrate some more boards to DSA, the lantiq/xrx200 target is 
using DSA in master now. It looks like some boards with qca8k would 
switch. These changes should be local to one target or even board 
anyway. The infrastructure is already provided. This can continue 
without much coordination and we can see what is finished when we branch.

Ansuel, 2021-09-30:
Hope i'm not polluting this... 
Anyway i made a parsing script to help devs convert to qca8k....
This is the output of the script run for ath79
https://pastebin.com/jCF0ziz4
If you find any qca83xx switch consider that most of the stuff is already handle and we can use the parsing script to convert the magic values... the script made it clear that we address 99% of the different config with only some strange undocumented stuff that will require an hack to set them...

abajk, 2022-01-05:
sunxi was done here: [47b531de](https://github.com/openwrt/openwrt/commit/47b531de3250de24158024a39beda21328f50701)

CHKDSK88, 2022-01-20:
Kirkwood was done: https://github.com/openwrt/openwrt/commit/4fd7e539e4f90128bdd7cb71c729a4b32f5de86e