aparcar/openwrt #15: Kernel 5.10

aparcar, 2021-09-29:
Targets using 5.10
- [ ] apm821xx
- [ ] arc770
- [ ] archs38
- [ ] armvirt
- [ ] at91
- [ ] ath25
- [ ] ath79
- [x] bcm27xx (via: https://github.com/openwrt/openwrt/commit/e1142041b0e65d9b20b65653fdb55d15abd37a42)
- [ ] bcm47xx
- [ ] bcm4908
- [x] bcm53xx (via: https://github.com/openwrt/openwrt/blob/master/target/linux/bcm53xx/Makefile)
- [ ] bcm63xx
- [x] bmips (via: https://github.com/openwrt/openwrt/commit/029093a302c9a66b74bec46285a179abd122a40a)
- [x] gemini (https://github.com/openwrt/openwrt/commit/f7e00d81bca720a5a5b073d614982b2f57f7da71)
- [ ] generic
- [ ] imx6
- [ ] ipq40xx
- [x] ipq806x (via: https://github.com/openwrt/openwrt/commit/0470159552641c2b11ccc1b0fcfcb4ea08f2c6ab)
- [ ] ipq807x
- [ ] kirkwood
- [ ] lantiq
- [ ] layerscape
- [ ] Makefile
- [ ] malta
- [ ] mediatek (via: https://github.com/openwrt/openwrt/commit/8f2cace0b7f2f9221ce3dd5a01772c17e5244c3e https://github.com/openwrt/openwrt/commit/c46ccb69d17e584479df849a107423175a143c83)
- [ ] mpc85xx
- [ ] mvebu
- [ ] mxs
- [ ] octeon
- [ ] octeontx
- [ ] omap
- [ ] oxnas
- [ ] pistachio
- [ ] ramips
- [ ] realtek
- [ ] rockchip
- [ ] sunxi
- [ ] tegra
- [ ] uml
- [ ] x86
- [ ] zynq

Notes from @hauke:

PolynomialDivision, 2021-10-01:
You linked the wrong PR for ipq40xx. ipq40xx is already accepted for testing:
https://github.com/openwrt/openwrt/pull/4548

CHKDSK88, 2021-10-03:
@yangbolu1991

Do You plan add 5.10 kernel support for layerscape?

aparcar, 2021-10-03:
This is merely an overview for myself

hauke, 2021-11-07:
archs38 was done here: da9a8fcbc3cf4319655da1b4d341134b5d4663c1
imx was done here: adc37b80406ab7ea0f8319ce8d81951731d8ee15
pistachio was done here: 646c011ec76eca151b24ccde01df12f47e199d2d
zynq was done here: 88537e76ec06012826d72e175b77c452cb41d256

uml is still on kernel 5.4 by default

aparcar, 2022-02-09:
Moved over to https://github.com/openwrt/openwrt/issues/5048

closed 2022-02-09 by aparcar